pulp-bio/wulpus #13: questions for using the gui for measuring A-mode and B-mode data

TimChang1024, 2024-09-04:
Solved:
•	3.2 Software Requirements
Unsolved:
•	2.4.2 Flashing the Device
Locate the .emProject file in the US_probe_nRF52_firmware folder as described and open it with Segger Embedded Studio.
Bottleneck: Unable to find the US_probe_nRF52_firmware folder.
•	2.5.1 Setting up the Toolchain
Install the Programmer application.
Bottleneck: The Programmer cannot be downloaded because the nRF Connect apps for Windows indicated in the manual are outdated.
Bottleneck of the Third Category:
•	The user manual does not specify how to power up the transmitter part. As a result, we cannot ensure the probe is working, and the nRF52 dongle is not connected to the probe.
•	If possible, we would appreciate a detailed, step-by-step procedure for the setup process, from start to successfully measuring the A-mode and B-mode graphs.

Sergio5714, 2024-12-17:
Thank you for your feedback. Indeed, there were a couple of inaccuracies. 
We updated the User Manual to address your comments. It will be available soon as a part of a new release.

Sergio5714, 2024-12-20:
We fixed this issue and merged it into the dev branch. The changes will be a part of the release 1.2.0.

closed 2024-12-20 by Sergio5714