pulp-bio/wulpus #18: Missing .epf File in Step 6 of Setup Instructions in page 16 of 2.3.2 Flashing the device(user manual)

EhsanNaeemi, 2024-11-21:
While following the setup instructions for the WULPUS repository, I encountered an issue at Step 6. 
The file **workspace_preferences.epf**, mentioned in the instructions, is missing from the repository. Without this file, it's impossible to complete the import and proceed with the setup.
Also, A search through the repository confirms that no .epf file exists.

Expected Behavior is:
The file workspace_preferences.epf should be available in the directory ultra-low-power-ultrasound-probe/dev_msp430/ to allow for importing preferences into CCS.

Actual Behavior is:
The file is missing, making it impossible to proceed beyond Step 6.

Thanks!

Sergio5714, 2024-11-25:
@EhsanNaeemi, thank you for submitting an issue! As discussed in private chat, the way to import the project is to go:

File->import project ->select the folder.
The CCS will automatically find the project.

We will update the description in a new release of the documentation (hopefully by the end of the year) together with other fixes.

Sergio5714, 2024-12-17:
Thank you again for your feedback! I addressed the issue in the version 1.1 of the User Manual, which will be a part of a new release.

Sergio5714, 2024-12-20:
We fixed this issue and merged the changes into the dev branch. The changes will be a part of the release 1.2.0.

closed 2024-12-20 by Sergio5714