pulp-bio/wulpus #19: Inquiry Regarding Connection Between Ultrasonic Transducer and Connector

Ethan010722, 2024-11-27:
Hello, Author,

First of all, thank you very much for sharing the ultrasonic collection system. I have successfully assembled most of the system. The only issue I am facing is figuring out how to connect the ultrasonic transducer to the connector. I am not entirely sure about the correct method for this connection and would appreciate any guidance or advice you could share.

Looking forward to your response. Thank you!

Sergio5714, 2024-12-03:
@Ethan010722, thank you for your question! Sorry for the delayed reply: we are currently quite busy and have delays in answering addressing the issues :)

Depending on the transducer, you might need a small adapter PCB to interface your transducer. On the HV PCB, we provide a board-to-wire connector. As outlined in the WULPUS User Manual, you would need the following components to interface it:

**Mating Connector and Pre-Crimped Wires:**
**Connector**: Hirose Electric DF52 Series
**Part Number:** DF52-16P-0.8C
**Description:** This is a 16-pin connector that mates with the WULPUS High-Voltage PCB's transducer port.

**Pre-Crimped Wires:**

**Part Number:** DF52-2832PF1571-28A9-300
**Description:** These pre-crimped wires are designed to work with the DF52 connector, enabling straightforward assembly of the transducer wiring.

For quick answers, we encourage you to try our ChatGPT bot that can help you to navigate the WULPUS User Manual:
https://chatgpt.com/g/g-XJs5YGjTh-wulpus-guide-assistant

Please let me know if I answered your question.

Ethan010722, 2024-12-03:
@Sergio5714 
Hi Sergio,

Thank you so much for your detailed response! I really appreciate your help. Your explanation and the provided information have resolved my issue.

Additionally, I would like to inquire if your ultrasonic system supports M-mode (Motion mode)?

Thank you again for your assistance!

Best regards,  
Ethan

Ethan010722, 2024-12-04:
[image]
[image]
Hello surgio,

I have connected the ultrasonic transducer to the board, but I don't know why there is almost no signal. I'm not sure where the problem lies, so I am seeking your advice.

Thank you very much for your help.

Best regards,
Ethan

Sergio5714, 2024-12-04:
Hello @Ethan010722 , Could you please redirect your questions to the following discussion?
https://github.com/pulp-bio/wulpus/discussions/20

I am closing this issue and will be happy to reply in aseparate discussion to keep the Issues cleaner.

closed 2024-12-04 by Sergio5714